UPB-PMRust/upb-pmrust #4: Write project rules and requirements

alexandruradovici, 2024-02-09:
- [ ] rules - what they can and can't do
- [ ] requirements - what are the project requirements

ZamfirAndreiPaul, 2024-02-10:
**Rules**

1. It is not allowed to use processors (eg Broadcom BCM2711, Broadcom BCM2712), you can only use microprocessors. Examples of microprocessors: RP2040, ATmega328, nRF52, PIC.
2. It is not allowed to copy both the code and the electronic schemes from the internet.
3. For the hardware part, both the breadboard and printed pcbs or test cables can be used.
4. Help from external sources is not allowed, the student can be asked by the coordinator about implementation details during the development period.
5. The student can ask questions to the coordinator during the development period.
6. Presentation of milestones is mandatory.
7. Projects must be accompanied by documentation, this should include project functionality, hardware development and software development.

alexandruradovici, 2024-02-10:
@ZamfirAndreiPaul pleae modify `docs/project.md` and open a PR.

DTeodor-Alexaandru, 2024-02-12:
> 1. It is not allowed to use processors (eg Broadcom BCM2711, Broadcom BCM2712), you can only use microprocessors. Examples of microprocessors: RP2040, ATmega328, nRF52, PIC.

I would change the words: processors  -> microprocessors and microprocessors -> microcontrollers
Also, I would add in the first rule a recomandation of using RP2040 because it's in the laboratory equipment.

Question: The project's software must be written in Rust or whatever they want? I think that by imposing them to use Rust will lessen the chances of plagiarism.

closed 2024-02-27 by alexandruradovici