UPB-PMRust/upb-pmrust #6: Establish the lab subjects for each week

alexandruradovici, 2024-02-09:
This is issue is opened to discuss the tab structure.

### Proposal

0. Introduction to Rust
    - Cheat Sheet Rust vs Java (they studied Java up to now)
    - Tour of Rust
1. Traits and generics
    - Tour of Rust
2. Setting up the Raspberry Pi Pico
    - Blink
   - Serial console
3. Debug
     - How to use gdb
4. GPIO
    - Pins 
    - Leds 
    - Buttons
5. Async/Await
6. PWM & ADC
7. Screen
8. Networking
    - WiFi
9. RTIC
10. Introduction to Tock
    - Flash OS / Flash Apps
    - Process Console
    - Tockloader
11. Tock Applications
    - Syscall Driver
12. Simple GPIO syscall driver
    - C API
    - Rust API

### Questions
1. I think we need to set up some labs for helping the students for the project
2. Do we need lab 1 *Trais and generics*, or can we merge it with lab 0, *Introduction to Rust*
3. What about swapping the Tock labs for the project?

DTeodor-Alexaandru, 2024-02-09:
We should consider 3-4 labs to help them with the project, 3 of them for milestones(doc, hardware, software). We can merge lab0 and lab1 and eliminate lab10, 11 and 12 to free up some space. 
Other solution would be to use just 1-2 labs for the project of which 1 hardware milestone and the other milestones to be assignments on git.

In the GPIO lab we can add pin interrupt, I think it will be useful for their project if we don't consider adding an Interrups lab.

alexandruradovici, 2024-02-09:
> In the GPIO lab we can add pin interrupt, I think it will be useful for their project if we don't consider adding an Interrups lab.

The idea of postponing interupts was to discuss async/await before them. But you might have a point. We could discuss interrupts first and use that information to explain async await.

alexandruradovici, 2024-02-10:
Merged GPIO with Interrupts

ZamfirAndreiPaul, 2024-02-10:
Laboratory 0 and 1 should be merged. Also, in lab 2, setting up the raspberry pi shouldn't take that long, that is, if the lab has 2 hours, in one hour we can discuss the raspberry pi pico, see what tools we use, mount the blinking led on the board and in the second hour we can look over the debugger, and that way we can do lab 2 and 3 and gain some space. And the part with TockOS, if it is taught at the end, will be difficult to integrate into the projects, because they don't have time in a week to integrate something, and probably, for them it will only be informative, it will not help them with the project. The idea of milestones seems very good to me, this way we see during the semester that they have progressed and they don't just come to the end with something done (copied), and they can't fool us like that, because we can ask questions at the moment presents. Otherwise, the structure seems ok to me, the question would be, if in the weeks with labs intended for the project, we call everyone to the laboratory, or only those who have questions. I would prefer to call them all so that we can see how they work and we can help them.

DTeodor-Alexaandru, 2024-02-15:
0. Introduction to Rust (this is the lab before the class)
    - Cheat Sheet Rust vs Java (they studied Java up to now)
    - Tour of Rust
1. Kicad
    - Load Pico schematic + footprint + 3D model
    - Make a simple schematic with LED + Button (source code available from us)
    - Measurements(need to see if we have enough multimeters)
2. Setting up the Raspberry Pi Pico + GPIO
    - Blink (DevBoard LED)
    - Serial console 
    - External LEDs + Buttons
3. Interrupts & Timers
    - PIN interrupts - 2 buttons - rise/fall edge change state of button
    - Timers - depending of timer interrupt change state of led -> blink
                  - use the 2 buttons to change the blink speed
4. PWM & ADC
    - ADC - read value from potentiometer
    - PWM - change frequency to see how led goes from blinking to "on" (use 2 buttons)
                - change duty cycle to see how led intensity changes (pot value from ADC)
5. Async/Await
    - try without tasking to do multiple thinks 
    - use async await for multitasking
6. SPI
    - Read/Write bits in an external memory
    - screen
7. I2C
    - 1 team master, 1 team slave
    - master team collects data, slave team controls a motor from collected data
8. Networking
    - WiFi
    - collect data from pico and send it to PC
    - select color from PC and modify RGB LED color
9. Help Project (Doc milestone)
10. Help Project 
11. Hardware milestone 
12. Software milestone

DTeodor-Alexaandru, 2024-02-16:
@ZamfirAndreiPaul @irina-nita @DanutAldea @irina-b-dev @alexandruradovici I need your feedback on the lab structure

ZamfirAndreiPaul, 2024-02-16:
For me looks very good.

alexandruradovici, 2024-02-19:
I opened issues for writing the labs.

alexandruradovici, 2024-02-19:
Do you think we should scatter the milestones instead of putting them at the end?

DTeodor-Alexaandru, 2024-02-19:
It wouldn't make much sense to scatter them. They first need to learn, then decide on what they can do and implement it. We can create small milestones for them to decide the project idea, to make an initial list of what components they need and so on... Those little milestones are mandatory but are not graded. We use them just to see if they really work on the project.

alexandruradovici, 2024-02-19:
That makes sense, the small milestones idea sounds good.

alexandruradovici, 2024-02-21:
For the courses, my suggestion is:

1. Introduction 
   - Overview of the course
   - Team
   - Rules
   - Project
   - AGC
   - How a processor works
2. The Cortex-M0+ Processor
    - Main components
    - How it boots
     - Memory Mapped I/O
    - GPIO
    - RP2040
3. Interrupts 
   - How interrupts work
   - Interrupts vs exceptions
   - Timers
4. PWM & ADC
   - Analog vs digital signals
   - Analog signal simulation (PWM)
   - ADC
 5. Asynchronous Development
    - async and await
    - How Rust works
 6. Data buses (UART, SPI)
 7. Data Buses (I2C, USB)
 8. Networking
    - Ethernet
    - BLE
    - WiFi
    - Thread
    - LoRa / LoRaWAN
 10. Memory Protection
     - MPU
     - MMU 
 11. RTOSes
     - What is real time?
     - How an RTOS works?
 12. Tock OS

DTeodor-Alexaandru, 2024-02-21:
Looks good to me. I would shift last 3 courses and add in week 9 a "Help project" course, Q&A. We can use this to help them with ideas, guide them if they have any problem hardware/software. I could come to the course to help out.

DTeodor-Alexaandru, 2024-02-22:
How about shifting the last 3 laboratories with a week? 
PM Fair would take place in the last week of the semester, and I wouldn't put the software milestone in the same week, to have an extra 3-4 days to work on the project to do small fixes.
The Documentation milestone can be checked by the laboratory assistant from home.

alexandruradovici, 2024-02-22:
Done