UPB-PMRust/upb-pmrust #84: PWM_Channels to PIN connection

NegrilaRares, 2024-03-29:
In the 4th lecture there is a table shown with possible PWM channels for the PINs on a RP2040. Where is says that all 30 GPIO pins on the RP2040 can be used for PWM but some of them don't accesable channels as shown below.

--> src\embassy\src\main.rs:47:9
    |
47  |         Pwm::new_output_a(peripherals.PWM_CH0, peripherals.PIN_1, config_green.clone());
    |         ^^^^^^^^^^^^^^^^^ the trait `PwmPinA<PWM_CH0>` is not implemented for `PIN_1`
    |
    = help: the following other types implement trait `PwmPinA<T>`:
              <PIN_0 as PwmPinA<PWM_CH0>>
              <PIN_2 as PwmPinA<PWM_CH1>>
              <PIN_4 as PwmPinA<PWM_CH2>>
              <PIN_6 as PwmPinA<PWM_CH3>>
              <PIN_8 as PwmPinA<PWM_CH4>>
              <PIN_10 as PwmPinA<PWM_CH5>>
              <PIN_12 as PwmPinA<PWM_CH6>>
              <PIN_14 as PwmPinA<PWM_CH7>>
            and 7 others 
            
            
            
[image]

NegrilaRares, 2024-03-29:
Although this may be just my lack of understanding on how to use 0A and 0B as channels.

alexandruradovici, 2024-04-01:
The `new_output_a` function creates a PWM for channel A of one of the PWM devices. 
In the data sheet, 0A, 1A, 2A etc mean PWM device 0, channel A, PWM device 1, channel A, PWM device 2, channel A ... . This means that Channel A  can have pins 0, 2, 4, ... .

The `PWM_CH0` type seems to be wrongly named, as it refers to the PWM device and not to one of the channels.

alexandruradovici, 2024-04-02:
Merged upstream, we renamed it to `PWM_SLICE`, the next release of embassy should have it fixed.

closed 2024-04-02 by alexandruradovici