UPB-PMRust/upb-pmrust #92: Unsafe toolbar

VladVulturescu, 2024-04-08:
While reading a course, when scrolling backwards too fast, it tries to call a previous slide which does not exist (index out of bounds). it results in a "page not found" issue. below are screenshots with the issue in the console. bug was discovered on macOS 12.3.1, but it can also be replicated on Windows and Linux

How to replicate bug: select a course, pick the last slide and scroll backwards fast

[image]
[image]
[image]

Authors: Vlad Vulturescu <vlad.vulturescu@yahoo.com> 
Andrei Dragota <anditu2003@gmail.com>

UdrAlberto, 2024-04-08:
I think the issue arises from the following js script
[image]

Specifically, the button triggers an event that requests the previous slide even if the current index points to the first position

alexandruradovici, 2024-04-08:
I not able to reproduce the bug. Please try the loading only the slides using https://embedded-rust-101.wyliodrin.com/slides/05. Does the bug still appear?

alexandruradovici, 2024-04-08:
The problem is from the combining of two projects: docusaurus for the website and slidev used in an iframe. Docusaurus rewrites links to use the react router, which intercept browser requests and stop slidev from working.

domnudragota, 2024-04-08:
Basically while reading the courses, we've clicked too fast going backwards. However, it was at random when this would trigger. We'll see if it works directly on slides.

VladVulturescu, 2024-04-08:
> I not able to reproduce the bug. Please try the loading only the slides using https://embedded-rust-101.wyliodrin.com/slides/05. Does the bug still appear?

I tried using the link provided to only load the slides and the same error occurred when clicking too fast.

[image]

alexandruradovici, 2024-04-08:
Are you clocking the browser's back button?

VladVulturescu, 2024-04-08:
No, I'm using the toolbar in the bottom left of the slide.

[image]

If it's relevant, I also managed to reproduce the bug by using the arrow keys on the keyboard. While testing to see if the bug occurs when using the arrow keys, I managed to make the bug occur in both current -> prev and current -> next ways, even at a moderate pace of pressing the keys.

UdrAlberto, 2024-04-08:
For me, the bug manifests either by what my colleague described or, more often, by getting a visual glitch that consists in multiple slides being over-layered

That is on Kali 2024.1 (Mercury browser) and Windows 11 build 22631 (Firefox browser)

[image]

alexandruradovici, 2024-05-05:
I think this might be a browser problem.

alexandruradovici, 2024-05-05:
Maybe it's worth submitting a bug to slide directly https://github.com/slidevjs/slidev, it seems to be a framework problem.

alexandruradovici, 2025-02-10:
We have updated slides, hopefully this fixes this.