UPB-PMRust/upb-pmrust #113: cache and volatile

jonathanpallant, 2024-04-23:
> read_volatile, write_volatile - no cache or compiler optimization

Unfortunately *volatile* isn't sufficient to bypass hardware caching. It's a compiler-level directive which tells the optimiser not to elide a read or write from an address because it may have a side effect that the compiler cannot see. Otherwise:

```rust
const UART_TX: *const u8 = 0xD000_0010;
for b in b"Hello, World".iter() {
    unsafe { UART_TX.write(*b); }
}
```

will optimise to

```rust
const UART_TX: *const u8 = 0xD000_0010;
unsafe { UART_TX.write(b'd'); }
```

because the later writes wrote to the same address as the earlier ones, and hence the earlier writes were useless. Using *volatile* says they are not useless.

Caching can only be disabled by programming the cache controller to mark specific regions of virtual address space as *uncacheable*. Fortunately, most Cortex-M systems don't have any cache - only the higher end Cortex-M7s tend to.

alexandruradovici, 2024-04-23:
I thought that using `volatile` in C/C++ is enough to prevent the processor to do caching. Do you know how this would be done in Rust? This is essential for MMIO.

jonathanpallant, 2024-04-23:
You have to talk to the cache controller to make the memory region uncacheable, or to flush the cache. Most chips set their I/O regions to be uncacheable by default, and RAM to be cacheable. Which is fine until you try and do some DMA, and the DMA can't see what's in the cache.

Or here is an example with a Cortex-A and a Cortex-M in the same CPU. Here the Cortex-M is flushing its cache so the Cortex-A can see what the Cortex-M just wrote.

https://github.com/thejpster/rust-beagleboardx15-demo/blob/166667f617ebb59f981361a48a5a3f52dd76cca3/bare-metal/ipu-demo/src/main.rs#L539

alexandruradovici, 2024-04-24:
I think I found the right wording, could you give us a review for #120? Thank you.

PR #120 (2024-04-25): Volatile is not the same as no cache
### Pull Request Overview

This pull request fixes the statement that `volatile` implies no cache. This pull request closes #113.

### Testing Strategy

This pull request was tested by reading the slides.

### TODO or Help Wanted

This pull request still needs feedback from @jonathanpallant.

### Build

- [x] Ran `npm build`.

### Author
closed 2024-04-25 by alexandruradovici